PaulStoffregen/XPT2046_Touchscreen #14: Fails with ESP32 - solution provided

hsgentry, 2018-03-26:
I have used this library many times with the ESP2866 with excellent results.  I recently ported a project to the ESP32.  It worked about 90% of the time, but occasionally it would crash with:"Guru Meditation Error: Core 1 panic'ed (Cache disabled but cached memory region accessed)".  I also use preferences.h to store data in NV ram.  The crash was caused by an interrupt from the touch screen while preferences was saving data to NV ram.  I added IRAM_ATTR  to the interrupt service routine in the .cpp file in the touch library.  

Changed:
 void  isrPin( void )

to:
 void IRAM_ATTR isrPin( void )

and the touch library is now 100% reliable on the ESP32.

The development environment is Arduino IDE 1.8.2 

touch library version 1.20

CelliesProjects, 2018-06-11:
I may have the same issue here. 
 `Guru Meditation Error: Core 1 panic'ed (Cache disabled but cached memory region accessed)`

I will try to your fix and report back.
I am at Arduino 1.8.5 though.

nrdmtt, 2019-11-11:
The same problem is present when this lib is used with an ESP8266.

The solution is to expand the #ifdef statement to 

```
#ifdef ESP32
void IRAM_ATTR isrPin( void )
#elif ESP8266
void ICACHE_RAM_ATTR isrPin ( void )
#else
void isrPin( void )
#endif
```