PaulStoffregen/XPT2046_Touchscreen #18: White screen issue with samd21

wajdib, 2018-10-07:
### Description

Using the example provided, I get white screen shortly after calling if (ts.touched())  on samd21 based boards(feather m0, arduino zero...)

### Steps To Reproduce Problem

run the example code.

### Hardware & Software

Board :feather m0

PaulStoffregen, 2018-10-07:
Which example?

This library comes with 3 examples.  2 of them do not control a display at all.  1 does, and the library it uses to do so is not compatible with those boards.

wajdib, 2018-10-07:
Touch test example.
My project already has TFT9341 library installed for graphics, once I add the following to my Loop function the screen goes all white:
`if (ts.touched()) {
}`

It seems that the library touched() function, which has an Update() function, its SPI calls interfere with the display and render it white.

PaulStoffregen, 2018-10-07:
Are you going to show the complete code to reproduce this problem?

wajdib, 2018-10-07:
Here is a complete to reproduce the problem:
`
#include <SPI.h>
#include <TFT9341.h>

#include "wiring_private.h"
#include "XPT2046_Touchscreen.h"

XPT2046_Touchscreen ts(12, 10); 

#if defined(ARDUINO_SAMD_ZERO) && defined(SERIAL_PORT_USBVIRTUAL)
  // Required for Serial on Zero based boards
  #define Serial SERIAL_PORT_USBVIRTUAL
#endif
//#endif
#define display Tft

void setup() {
 
  ts.begin();
  Serial.begin(115200);
 
display.InitLCD(3);
  //set background color
   display.fillScr(0, 255, 0);
  //display.clrScr();

}

float speed = 0;
int OldOnes = 0;
int OldTens = 0;
extern uint8_t SevenSeg_XXXL_Num[];
void loop() {

//If ts.touched() is commented out, you will see the counter properly updating on the screen,
//if ts.touched() is uncommented, nothing will be displayed.
if (ts.touched()) {
    }
 
  delay(1000);
    speed++;

  display.setBackColor(0x0000);
  display.setColor(0xFFFF);
  display.setFont(SevenSeg_XXXL_Num);

  

  int ones = (int)speed % 10;
  int tens = (int)speed / 10 % 10;

  if (OldOnes != ones )
  {
    // display.setTextSize(2);
    display.setColor(0xFFFF);
    OldOnes = ones;
    String sOnes = String(ones);
    if (ones == 0)
      sOnes = "0";
    display.print(sOnes, 155, 70);
  }
  if (OldTens != tens )
  {
    display.setColor(0xFFFF);
    OldTens = tens;
    String sTens = String(tens);
    if (tens == 0)
      sTens = "0";
    display.print(sTens, 95, 70);
  }

 
}`

wajdib, 2018-10-07:
TFT9341 library Im using is this one https://github.com/pantata/TFT9341SPI