PaulStoffregen/XPT2046_Touchscreen #26: XPT2046_Touchscreen not working with Adafruit_ILI9341 li

sonastyle, 2019-08-12:
### Description

Describe your problem.

### Steps To Reproduce Problem

Please give detailed instructions needed for anyone to attempt to reproduce the problem.

### Hardware & Software

Board
Shields / modules used
Arduino IDE version
Teensyduino version (if using Teensy)
Version info & package name (from Tools > Boards > Board Manager)
Operating system & version
Any other software or hardware?

### Arduino Sketch

demonstrates the problem)
#include <Adafruit_ILI9341.h>
#include <SPI.h>
#include <Adafruit_GFX.h>
#include <XPT2046_Touchscreen.h>

#define TS_MINX 400
#define TS_MINY 330
#define TS_MAXX 3875
#define TS_MAXY 3700

// For the Adafruit shield, these are the default.
#define TFT_DC 26
#define TFT_CS 5
#define TFT_MOSI  23
#define TFT_CLK	  18
#define TFT_RST	  -1
#define TFT_MISO  19
#define Touch_CS 22
#define Touch_IRQ 21

#define YP TFT_RST  // must be an analog pin, use "An" notation!	   Y+	//cs
#define XM TFT_RST  // must be an analog pin, use "An" notation!	   X-	//cd
//#define YP TFT_CS  // must be an analog pin, use "An" notation!	   Y+	//cs
//#define XM TFT_DC  // must be an analog pin, use "An" notation!	   X-	//cd
#define YM 0   // can be a digital pin						   Y-
#define XP 0   // can be a digital pin						   X+

#define BLACK   0x0000
#define BLUE    0x001F
#define RED     0xF800
#define GREEN   0x07E0
#define CYAN    0x07FF
#define MAGENTA 0xF81F
#define YELLOW  0xFFE0
#define WHITE   0xFFFF

const unsigned char circle[]PROGMEM = {
0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,
0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,
0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,
0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,
0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,
0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,
0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,
0x0,0x0,0x0,0x7,0xf8,0x0,0x0,0x0,0x0,
0x0,0x0,0x0,0x7f,0xff,0x80,0x0,0x0,0x0,
0x0,0x0,0x3,0xff,0xff,0xf0,0x0,0x0,0x0,
0x0,0x0,0xf,0xff,0xff,0xfc,0x0,0x0,0x0,
0x0,0x0,0x1f,0xff,0xff,0xfe,0x0,0x0,0x0,
[... 729 lines cut ...]
if (board[2] == 2 && board[4] == 2 && board[6] == 2) {
winner = 2;
}

}
void drawBitmap(int16_t x, int16_t y,
const uint8_t *bitmap, int16_t w, int16_t h, uint16_t color) {

int16_t i, j, byteWidth = (w + 7) / 8;
uint8_t byte;

for (j = 0; j<h; j++) {
for (i = 0; i<w; i++) {
if (i & 7) byte <<= 1;
else      byte = pgm_read_byte(bitmap + j * byteWidth + i / 8);
if (byte & 0x80) tft.drawPixel(x + i, y + j, color);
}
}
}

#Hardware
ESP32 wroom chip with Arduino IDE 1.8.2 and ESP32 Arduino SDK 1.0.2

### Errors or Incorrect Output
Display stops working after touch screen spi begin.

Querela, 2020-11-12:
I still have the same problem. Maybe with a more canonical example: XPT2046_Calibrate library/example sketches. Same initialization as used above. Same hardware setup.

The touchscreen works until I call `ts.begin()` and then nothing. Before I can draw, afterwards I can only receive touch events. If I call `tft.begin()` after `ts.begin()` I still can't draw.
I would not think that it is an issue with different SPI Settings but anything else seems also unlikely? Alternatively, I can only use 3 more pins for separate SPI lines but that's only a stop-gap solution.

PaulStoffregen, 2020-11-12:
Can you test with a non-ESP board?

We've had bug reports on Ethernet which only happen with ESP32.  I'm starting to suspect their SPI library may have bugs...

Querela, 2020-11-12:
Mhh. Ok. I tested some other things.
I found that it works, if I do not specify the SPI Pins in the Adafruit library (I used the defaults, but without specifying RESET it did not work ...)

So what worked:
```cpp
#define TFT_VCC -1 // 3.3V
#define TFT_GND -1 // opposite to 3.3V
#define TFT_CS 5 // Chip Select
#define TFT_RST 17 // RESET
#define TFT_DC 16
#define TFT_MOSI 23 // Master Out Slave In
#define TFT_CLK 18 // SCK (Clock)
#define TFT_LED -1 // 5V (Backlighting)
#define TFT_MISO 19 // Master In Slave Out

//#define T_CLK TFT_CLK
#define T_CS 26
//#define T_DIN TFT_MOSI
//#define T_DO TFT_MISO
#define T_IRQ 25

// -------------------------------------

/* TFT did not work as it did not have a correct RESET pin, I guess */
// Adafruit_ILI9341 tft = Adafruit_ILI9341(TFT_CS, TFT_DC);
/* TFT worked, but not in combination with Touch; SPI-Pins are standard SPI pins on ESP32-devkit C */
// Adafruit_ILI9341 tft = Adafruit_ILI9341(TFT_CS, TFT_DC, TFT_MOSI, TFT_CLK, TFT_RST, TFT_MISO);
/* this seems to work as it still uses the same SPI pins as above but implicitely somehow */
Adafruit_ILI9341 tft = Adafruit_ILI9341(TFT_CS, TFT_DC, TFT_RST);

// Touch Calibration (is a fork on your library, essentially the same)
XPT2046_Calibrated ts(T_CS, T_IRQ);
```

Before, I was thinking on extending your constructor to accept custom SPI pins ...

ccvelandres, 2020-12-08:
I think this has to do with SPI.begin() in XPT2046_Touchscreen::begin() overriding the current setup for the SPI port. Same for the SPI_SETTING macro which I think persists after the transaction used by this library. This breaks the expected SPI configuration which the other libraries are expecting.

I removed the SPI.begin() call from XPT2046_Touchscreen::begin() and replaced the SPI_SETTING to blank. This fixed my problem with compatibility with other libraries which uses the same SPI port. I think you need to add an option to remove the SPI initialization from this library.

wfdudley, 2021-06-15:
I have a short test program, which, when compiled for and run on an Arduino Mega 2560,
exhibits this behaviour.  With "#define TOUCH 1", tft writes fail to display on the TFT.
when TOUCH is set to 0 (i.e. your library code disabled), tft writes work correctly.
Arduino 1.8.13.  Latest version of Adafruit_ILI9341 and Adafruit_GFX_Library (whatever
they are).  Example code is 109 lines including comments.
[bug_9341_2046.zip](https://github.com/PaulStoffregen/XPT2046_Touchscreen/files/6659105/bug_9341_2046.zip)

alexmogavero, 2021-11-03:
> Mhh. Ok. I tested some other things. I found that it works only if I do not specify the SPI pins in the Adafruit library (I used the defaults, but without specifying RESET it did not work, so I started with specifying all pins)
> 
> So what worked:
> 
> ```c++
> #define TFT_VCC -1 // 3.3V
> #define TFT_GND -1 // opposite to 3.3V
> #define TFT_CS 5 // Chip Select
> #define TFT_RST 17 // RESET
> #define TFT_DC 16
> #define TFT_MOSI 23 // Master Out Slave In
> #define TFT_CLK 18 // SCK (Clock)
> #define TFT_LED -1 // 5V (Backlighting)
> #define TFT_MISO 19 // Master In Slave Out
> 
> //#define T_CLK TFT_CLK
> #define T_CS 26
> //#define T_DIN TFT_MOSI
> //#define T_DO TFT_MISO
> #define T_IRQ 25
> 
> // -------------------------------------
> 
> /* TFT did not work as it did not have a correct RESET pin, I guess */
> // Adafruit_ILI9341 tft = Adafruit_ILI9341(TFT_CS, TFT_DC);
> /* TFT worked, but not in combination with Touch; SPI-Pins are standard SPI pins on ESP32-devkit C */
> // Adafruit_ILI9341 tft = Adafruit_ILI9341(TFT_CS, TFT_DC, TFT_MOSI, TFT_CLK, TFT_RST, TFT_MISO);
> /* this seems to work as it still uses the same SPI pins as above but implicitely somehow */
> Adafruit_ILI9341 tft = Adafruit_ILI9341(TFT_CS, TFT_DC, TFT_RST);
> 
> // Touch Calibration (is a fork of your library, essentially the same in how to connects via SPI)
> XPT2046_Calibrated ts(T_CS, T_IRQ);
> ```
> 
> Before, I was thinking on extending your constructor to accept custom SPI pins ...

I can confirm this worked for me. I also tried to specify all the PINs, but despite the pin number has not changed, it was not working and now it is working. Very odd indeed!

clutch2sft, 2023-07-30:
@Querela  Saved my bacon! I had same behavoir on an M0 board.  Could not get display to work using:

//Adafruit_ILI9341 tft = Adafruit_ILI9341(TFT_CS, TFT_DC);

Some sort of conflict creation happens using this:

Using This:

//ILI9341Display() : tft(TFT_CS, TFT_DC, TFT_MOSI, TFT_CLK, TFT_RST, TFT_MISO) { }

Everything is happy using this:

ILI9341Display() : tft(TFT_CS, TFT_DC, TFT_RST) { }

Thank you for this thread!!!  Maybe someone can figure out what happens to the SPI bus when you pass specific pins to ILI9341 driver.

NAMost, 2023-08-06:
Thank you, you've saved my times. The trick is to use the simple constructor.