PaulStoffregen/XPT2046_Touchscreen #29: Redirect ZPT2046 to another SPI hardware or SPI software port

Jim-bee, 2020-01-02:
Please use this form only to report code defects or bugs.

For any question, even questions directly pertaining to this code, post your question on the forums related to the board you are using.

Arduino: forum.arduino.cc
Teensy: forum.pjrc.com
ESP8266: www.esp8266.com
ESP32: www.esp32.com
Adafruit Feather/Metro/Trinket: forums.adafruit.com
Particle Photon: community.particle.io

If you are experiencing trouble but not certain of the cause, or need help using this code, ask on the appropriate forum.  This is not the place to ask for support or help, even directly related to this code.  Only use this form you are certain you have discovered a defect in this code!

Please verify the problem occurs when using the very latest version, using the newest version of Arduino and any other related software.

----------------------------- Remove above -----------------------------

### Description

Describe your problem.

### Steps To Reproduce Problem

Please give detailed instructions needed for anyone to attempt to reproduce the problem.

### Hardware & Software

Board
Shields / modules used
Arduino IDE version
Teensyduino version (if using Teensy)
Version info & package name (from Tools > Boards > Board Manager)
Operating system & version
Any other software or hardware?

### Arduino Sketch

```cpp
// Change the code below by your sketch (please try to give the smallest code which demonstrates the problem)
#include <Arduino.h>

// libraries: give links/details so anyone can compile your code for the same result

void setup() {
}

void loop() {
}
```

### Errors or Incorrect Output

If you see any errors or incorrect output, please show it here.  Please use copy & paste to give an exact copy of the message.  Details matter, so please show (not merely describe) the actual message or error exactly as it appears.

PaulStoffregen, 2021-06-27:
This is solved, right?

Jim-bee, 2021-06-28:
It can be closed.  Thanks.

closed 2021-06-28 by PaulStoffregen