PaulStoffregen/XPT2046_Touchscreen #30: Compile ERROR  with ESP8266(NodeMCU-1.0(ESP-12E Module)

2969773606, 2020-01-06:
Compile ERROR:

D:\Documents\Arduino\libraries\ILI9341_t3-master/ILI9341_t3.h: In member function 'void ILI9341_t3::waitFifoNotFull()':

D:\Documents\Arduino\libraries\ILI9341_t3-master/ILI9341_t3.h:445:9: error: 'KINETISK_SPI0' was not declared in this scope

    sr = KINETISK_SPI0.SR;

         ^

D:\Documents\Arduino\libraries\ILI9341_t3-master/ILI9341_t3.h: In member function 'void ILI9341_t3::waitFifoEmpty()':

D:\Documents\Arduino\libraries\ILI9341_t3-master/ILI9341_t3.h:453:9: error: 'KINETISK_SPI0' was not declared in this scope

    sr = KINETISK_SPI0.SR;

2969773606, 2020-01-06:
Thank you very much for any advice.

FrankBoesing, 2020-01-06:
Ili9341_t3 is for Teensy 3, only. No Chance to use it with ESP.

PaulStoffregen, 2020-01-06:
Of course ILI9341_t3 doesn't compile for ESP.  It's a library for Teensy.

Why you're reporting ILI9341_t3 on this library's repository makes no sense.

closed 2020-01-06 by PaulStoffregen

2969773606, 2020-01-06:
Thank you very much, it make sense.

2969773606, 2020-01-06:
> Ili9341_t3 is for Teensy 3, only. No Chance to use it with ESP.

Thank you very much, it make sense.

2969773606, 2020-01-06:
> Of course ILI9341_t3 doesn't compile for ESP. It's a library for Teensy.
Thank you very much, it makes sense.