PaulStoffregen/XPT2046_Touchscreen #31: esp32 modification needed for 2.4" touchscreen

jponko, 2020-02-15:
### Description

Describe your problem.

Touchscreen y-coordinate is reversed in when display is in landscape mode. The solution is to add another case to void XPT2046_Touchscreen::update() in XPT2046_Touchscreen.cpp atter line 170 such as:
          case 3:
            xraw = x;
            yraw = 4095 - y;
            break;
And then modify XPT2046_Touchscreen.h line 53 to be:
void setRotation(uint8_t n) { rotation = n % 5; }

### Steps To Reproduce Problem

Please give detailed instructions needed for anyone to attempt to reproduce the problem.

in Linux OS:
1) install Arduino IDE 1.8.11
2) add Adafruit GFX(1.7.5), Adafruit ILI9341(1.5.4) and XPT_touchscreen(1.3.0) libraries through Library Manager,
3) install esp32 board support  to IDE in prefs https://dl.espressif.com/dl/package_esp32_index.json
4) git clone https://github.com/erikkaashoek/tinySA to your prefered directory.
5) Wire display to ESP32 and plug USB cable from computer into ESP32.
6) Open tinySA.ino in Arduino IDE and under Tools dropdown select "DOIT ESP32 DEVKIT V1" as your 
board. and 921600 as upload speed. 
7) Upload sketch and touch display to display menu. Attempt to touch a menu item. The y screen          coordinate will not match menu item location.
----------------- HARDWARE & SOFTWARE -----------------------------
Board: ESP32 Doit DevkitV1 board
TFT: 2.4" noname tft 320x240 resistive touchscreen as shown on this git.
Arduino IDE version 1.8.11
Version info & package name (from Tools > Boards > Board Manager) :ESP32 1.0.4
Operating system & version: Linux Mint 19.2
Any other software or hardware? USB mini cable

### Arduino Sketch
see above github location

// libraries: give links/details so anyone can compile your code for the same result
see above

tonhuisman, 2020-09-18:
> Touchscreen y-coordinate is reversed in when display is in landscape mode.

What I found on my ILI9341/XPT2046 combo is that the touchscreen seems to be mounted 180 degrees rotated, so when switching to landscape mode, the TFT is rotated by 270 degrees, and the touchscreen needs to be rotated 90 degrees (as the default/0 degrees is portrait). I assume that is caused by 'Chinese manufacturing'.

> `void setRotation(uint8_t n) { rotation = n % 5; }`

Well, in fact that should be: `void setRotation(uint8_t n) { rotation = n % 4; }` as there are **4** possible values (0..3)

jponko, 2020-09-18:
"I assume that is caused by 'Chinese manufacturing'." 
 
You're probably correct about the manufacturing problem but there isn't much that's not manufactured in China.

closed 2020-09-18 by jponko