PaulStoffregen/XPT2046_Touchscreen #35: The Version 1.3 has not the ESP8266 fix

GerLech, 2020-11-18:
Please add the ESP8266 fix to a new version

Derek-K, 2021-06-26:
Not sure what "fix" you are referring to, but I just tested this on my ESP8266 and it works fine.

Otherwise "TouchTest" example works for me with a slight change (because my board doesn't even have a header for the IRQ output.

```
#define CS_PIN  D1
// MOSI=11, MISO=12, SCK=13

XPT2046_Touchscreen ts(CS_PIN, 255);  // Param 2 - 255 - No interrupts
```

p.s. It needs TFT_eSPI, please check setup.h to config for your LCD (which is not related to XPT2046)

```
#include <XPT2046_Touchscreen.h>
#include <SPI.h>
#include <TFT_eSPI.h> // Hardware-specific library
#include <SPI.h>

#define CS_PIN  D1
// MOSI=11, MISO=12, SCK=13

//XPT2046_Touchscreen ts(CS_PIN);
#define TIRQ_PIN  2
//XPT2046_Touchscreen ts(CS_PIN);  // Param 2 - NULL - No interrupts
XPT2046_Touchscreen ts(CS_PIN, 255);  // Param 2 - 255 - No interrupts
//XPT2046_Touchscreen ts(CS_PIN, TIRQ_PIN);  // Param 2 - Touch IRQ Pin - interrupt enabled polling
TFT_eSPI tft = TFT_eSPI();       // Invoke custom library

void setup() {
  Serial.begin(115200);
  Serial.println("Start");
  ts.begin();
  ts.setRotation(1);
  tft.init();
  tft.setRotation(0);
  tft.fillScreen(ILI9341_BLACK);
  Serial.println("Init Done");
  //while (!Serial && (millis() <= 1000));
}
boolean wastouched = true;
void loop() {
  boolean istouched = ts.touched();
  if (istouched) {
    TS_Point p = ts.getPoint();
    if (!wastouched) {
      tft.fillScreen(ILI9341_BLACK);
      tft.setTextColor(ILI9341_YELLOW);
      tft.setTextFont(6);
      tft.setCursor(60, 80);
      tft.print("Touch");
    }
    tft.fillRect(100, 150, 140, 60, ILI9341_BLACK);
    tft.setTextColor(ILI9341_GREEN);
    tft.setTextFont(4);
    tft.setCursor(100, 150);
    tft.print("X = ");
    tft.print(p.x);
    tft.setCursor(100, 180);
    tft.print("Y = ");
    tft.print(p.y);
    Serial.print(", x = ");
    Serial.print(p.x);
    Serial.print(", y = ");
    Serial.println(p.y);
  } else {
    if (wastouched) {
      tft.fillScreen(ILI9341_BLACK);
      tft.setTextColor(ILI9341_RED);
      tft.setTextFont(4);
      tft.setCursor(120, 50);
      tft.print("No");
      tft.setCursor(80, 120);
      tft.print("Touch");
    }
    Serial.println("no touch");
  }
  wastouched = istouched;
  delay(100);
}
```

GerLech, 2021-06-26:
I  refer the following fix:
https://github.com/PaulStoffregen/XPT2046_Touchscreen/commit/26b691b2c84ad93884e262088d3509bc17241278#diff-98addf8b1b200ee0c8537d66219923d082a3dc78e45b58029373aaf50593e3d1
It was committed on 25 Jun 2019 but the release 1.3 was already released 15 Feb 2019, so it can not contain the referred fix. If I update the source files to the last commited version, it is fine, but if you install the library from Arduino library manager, version 1.3 is used which does not contain the fix.

Derek-K, 2021-06-27:
I see... it is more specific to the interrupt issue, sorry, couldn't read your mind earlier with your original post.

PaulStoffregen, 2021-06-27:
https://github.com/PaulStoffregen/XPT2046_Touchscreen/releases/tag/v1.4

closed 2021-06-27 by PaulStoffregen