PaulStoffregen/XPT2046_Touchscreen #36: missing: <font_Arial.h> // from ILI9341_t3

Greece2, 2020-12-15:
The #include <font_Arial.h> // from ILI9341_t3 seems not to be working.
ILI9341_t3 is installed, but that one only comes with glcdfont.c. Where to find the font_Arial.h ?

Tried to compile the ILI9341Test from the examples for this lib.

Thanks in advance !

FrankBoesing, 2020-12-15:
1.  This is the XPT2046 Library - it has nothing to do with the ILI9341\_t3 lib :)
2.  The font is included in the ILI9341 Library. See here: https://github.com/PaulStoffregen/ILI9341_t3 If you don't have it, you sis something wrong with the installation.

PaulStoffregen, 2020-12-15:
Quick answer: install the latest Teensyduino.  Then use File > Prefs to turn on verbose output.  Watch for the message about duplicate libs.  Then click Verify.  When Arduino tells you the path of the old copy it's using, find and delete that old copy.

Longer answer: ask on the forum if you need more help.  These github issues aren't the place for questions.

closed 2020-12-15 by PaulStoffregen

Greece2, 2020-12-15:
Thanks for all your answers! 
It seems the ILI9341_t3 was incomplete. Downloaded it as ZIP and reinstalled, works fine now.
Sorry if I have ask in the wrong place and the wrong question.  Won't ask again.