PaulStoffregen/XPT2046_Touchscreen #38: [not working!!!] Header File missing for ILI9341test!

hasenradball, 2021-04-09:
Have you ever tested your own deliverd examples????

### Description

Header File missing for ILI9341test!

### Steps To Reproduce Problem

Please give detailed instructions needed for anyone to attempt to reproduce the problem.

### Hardware & Software

Board
Shields / modules used
Arduino IDE version
Teensyduino version (if using Teensy)
Version info & package name (from Tools > Boards > Board Manager)
Operating system & version
Any other software or hardware?

### Arduino Sketch

```cpp
// Change the code below by your sketch (please try to give the smallest code which demonstrates the problem)
#include <Arduino.h>

// libraries: give links/details so anyone can compile your code for the same result

void setup() {
}

void loop() {
}
```

### Errors or Incorrect Output

If you see any errors or incorrect output, please show it here.  Please use copy & paste to give an exact copy of the message.  Details matter, so please show (not merely describe) the actual message or error exactly as it appears.

Derek-K, 2021-06-26:
That's not directly related to the XPT2046 library, you need to download ILI9341 library 
(depends on your IDE, for example with Arduino IDE, go to Library Manager, search for ILI9341_t3 and install it)

PaulStoffregen, 2021-06-28:
One of the 3 examples in this library depends upon another library.

Not a bug.

closed 2021-06-28 by PaulStoffregen