PaulStoffregen/XPT2046_Touchscreen #45: DOCUMENTATION ON SPI NOT CLEAR

amitsamanta996, 2022-06-30:
----------------------------- Remove above -----------------------------

### Description

DOCUMENTATION ON SPI NOT CLEAR

### Steps To Reproduce Problem

Documentation on how to use the SPIClass for different spi pins is not mentioned but code is present

### Hardware & Software

Board ESP32
Shields / modules used
Arduino IDE version 1.8.19
Teensyduino version (if using Teensy)
Version info & package name (from Tools > Boards > Board Manager)  ESP32 DEVKIT V1
Operating system & version win11
Any other software or hardware?

### Arduino Sketch

```cpp
// Change the code below by your sketch (please try to give the smallest code which demonstrates the problem)
#include <Adafruit_GFX.h>      
#include <Adafruit_ILI9341.h>   
#include <Fonts/FreeSerif12pt7b.h>
#include <XPT2046_Touchscreen.h>
#include <SPI.h>

// libraries: give links/details so anyone can compile your code for the same result

#define TFT_SCK    14
#define TFT_MOSI   13
#define TFT_MISO   12
#define TFT_CS    27   
#define TFT_DC    26     
#define CS_PIN    25
Adafruit_ILI9341 tft = Adafruit_ILI9341(TFT_CS, TFT_DC, TFT_MOSI, TFT_SCK, -1, TFT_MISO); 
XPT2046_Touchscreen ts(CS_PIN);

void setup() {
tft.begin();
  tft.setRotation(3);
  ts.begin();   //how to pass SPICLass here??????
  ts.setRotation(1);
}

void loop() {
}
```

### Errors or Incorrect Output

If you see any errors or incorrect output, please show it here.  Please use copy & paste to give an exact copy of the message.  Details matter, so please show (not merely describe) the actual message or error exactly as it appears.

ErikDorstel, 2023-03-18:
After reading the source code I found a solution to use HSPI instead of VSPI.
[ili9341esp32.txt](https://github.com/PaulStoffregen/XPT2046_Touchscreen/files/11009330/ili9341esp32.txt)

ErikDorstel, 2023-03-18:
After reading the source code I found a solution to use HSPI instead of VSPI.

#include <SPI.h>
#include <Adafruit_GFX.h>
#include <Adafruit_ILI9341.h>
#include <XPT2046_Touchscreen.h>

#define TFT_CS 15
#define TFT_DC 17
#define TFT_MOSI 13
#define TFT_CLK 14
#define TFT_RST 4
#define TFT_MISO 12
#define Touch_CS 33
#define Touch_IRQ 34

SPIClass *hspi=new SPIClass();

Adafruit_ILI9341 tft=Adafruit_ILI9341(hspi,TFT_DC,TFT_CS,TFT_RST);
XPT2046_Touchscreen ts(Touch_CS,Touch_IRQ);

void setup() {
  hspi->begin(TFT_CLK,TFT_MISO,TFT_MOSI);
  tft.begin();
  tft.setRotation(3);
  tft.fillScreen(ILI9341_BLACK);
  tft.setTextColor(ILI9341_WHITE,ILI9341_BLACK);
  tft.setTextSize(2);
  tft.setCursor(0,0);
  tft.println("ILI9341 + XPT2046 Test");
  ts.begin(*hspi);
  ts.setRotation(1); }

void loop() {
  if (ts.touched()) {
    TS_Point touch=ts.getPoint();
    tft.setCursor(50,100);
    tft.print("X:"); tft.print(touch.x); tft.print("  ");
    tft.print("Y:"); tft.print(touch.y); tft.println("      "); } }

ErikDorstel, 2023-03-18:
After reading the source code I found a solution to use HSPI instead of VSPI.

`#include <SPI.h>
#include <Adafruit_GFX.h>
#include <Adafruit_ILI9341.h>
#include <XPT2046_Touchscreen.h>

#define TFT_CS 15
#define TFT_DC 17
#define TFT_MOSI 13
#define TFT_CLK 14
#define TFT_RST 4
#define TFT_MISO 12
#define Touch_CS 33
#define Touch_IRQ 34

SPIClass *hspi=new SPIClass();

Adafruit_ILI9341 tft=Adafruit_ILI9341(hspi,TFT_DC,TFT_CS,TFT_RST);
XPT2046_Touchscreen ts(Touch_CS,Touch_IRQ);

void setup() {
  hspi->begin(TFT_CLK,TFT_MISO,TFT_MOSI);
  tft.begin();
  tft.setRotation(3);
  tft.fillScreen(ILI9341_BLACK);
  tft.setTextColor(ILI9341_WHITE,ILI9341_BLACK);
  tft.setTextSize(2);
  tft.setCursor(0,0);
  tft.println("ILI9341 + XPT2046 Test");
  ts.begin(*hspi);
  ts.setRotation(1); }

void loop() {
  if (ts.touched()) {
    TS_Point touch=ts.getPoint();
    tft.setCursor(50,100);
    tft.print("X:"); tft.print(touch.x); tft.print("  ");
    tft.print("Y:"); tft.print(touch.y); tft.println("      "); } }`

ErikDorstel, 2023-03-18:
After reading the source code I found a solution to use HSPI instead of VSPI.

```c
#include <SPI.h>
#include <Adafruit_GFX.h>
#include <Adafruit_ILI9341.h>
#include <XPT2046_Touchscreen.h>

#define TFT_CS 15
#define TFT_DC 17
#define TFT_MOSI 13
#define TFT_CLK 14
#define TFT_RST 4
#define TFT_MISO 12
#define Touch_CS 33
#define Touch_IRQ 34

SPIClass *hspi=new SPIClass();

Adafruit_ILI9341 tft=Adafruit_ILI9341(hspi,TFT_DC,TFT_CS,TFT_RST);
XPT2046_Touchscreen ts(Touch_CS,Touch_IRQ);

void setup() {
  hspi->begin(TFT_CLK,TFT_MISO,TFT_MOSI);
  tft.begin();
  tft.setRotation(3);
  tft.fillScreen(ILI9341_BLACK);
  tft.setTextColor(ILI9341_WHITE,ILI9341_BLACK);
  tft.setTextSize(2);
  tft.setCursor(0,0);
  tft.println("ILI9341 + XPT2046 Test");
  ts.begin(*hspi);
  ts.setRotation(1); }

void loop() {
  if (ts.touched()) {
    TS_Point touch=ts.getPoint();
    tft.setCursor(50,100);
    tft.print("X:"); tft.print(touch.x); tft.print("  ");
    tft.print("Y:"); tft.print(touch.y); tft.println("      "); } }
```

PaulStoffregen, 2023-03-18:
Use of alternate SPI ports is shown in these examples.

https://github.com/PaulStoffregen/XPT2046_Touchscreen/blob/cf3364816fd49c086b66fd7d0bb99fb270722155/examples/TouchTest/TouchTest.ino#L16

https://github.com/PaulStoffregen/XPT2046_Touchscreen/blob/cf3364816fd49c086b66fd7d0bb99fb270722155/examples/TouchTestIRQ/TouchTestIRQ.ino#L14

PaulStoffregen, 2023-03-18:
I have also added it to the readme file

https://github.com/PaulStoffregen/XPT2046_Touchscreen/commit/5d5120e93ab5d28b3f1db6d754819c354c2da019

I believe this should fully satisfy any concern whether use of other SPI ports is documented.

closed 2023-03-18 by PaulStoffregen