PaulStoffregen/XPT2046_Touchscreen #46: touchscreen just display wrong data with ili9488

cho934, 2023-03-23:
I'm using teensy 4.0, and I connected only the touchscreen of the LCD 3.5" v1.0  (ili9488) by SPI in order to have a simple test.

I tested both examples with IRQ and without, in both cases, Data on the serial are displays strange number even if I didn't touch the screen.

```
3:22:44:     z1=0,z2=0      z=4095 p=4095,  0,0  0,0  0,0Pressure = 4095, x = 0, y = 0
23:22:45:     z1=0,z2=0      z=4095 p=4095,  0,0  0,0  0,0Pressure = 4095, x = 0, y = 0
23:22:45:     z1=0,z2=0      z=4095 p=4095,  0,0  0,0  0,0Pressure = 4095, x = 0, y = 0
23:22:45:     z1=0,z2=0      z=4095 p=4095,  0,0  0,0  0,0Pressure = 4095, x = 0, y = 0
23:22:45:     z1=0,z2=0      z=4095 p=4095,  0,0  0,0  0,0Pressure = 4095, x = 0, y = 0
23:22:45:     z1=24,z2=18      z=4101 p=4101,  0,0  0,0  0,0Pressure = 4101, x = 0, y = 0
23:22:45:     z1=0,z2=0      z=4095 p=4095,  0,0  0,0  0,0Pressure = 4095, x = 0, y = 0
23:22:45:     z1=24,z2=18      z=4101 p=4101,  26,18  26,18  26,0Pressure = 4101, x = 26, y = 18
23:22:45:     z1=0,z2=0      z=4095 p=4095,  0,0  0,0  0,0Pressure = 4095, x = 0, y = 0
23:22:45:     z1=24,z2=18      z=4101 p=4101,  26,18  26,18  26,0Pressure = 4101, x = 26, y = 18
23:22:45:     z1=0,z2=0      z=4095 p=4095,  0,0  0,0  0,0Pressure = 4095, x = 0, y = 0
23:22:45:     z1=24,z2=18      z=4101 p=4101,  26,18  26,18  26,0Pressure = 4101, x = 26, y = 18
23:22:45:     z1=0,z2=0      z=4095 p=4095,  0,0  0,0  0,0Pressure = 4095, x = 0, y = 0
```

If I tried exactly the same code with another LCD 2.8" v1.2 (ili9341), It works perfectly !

What's wrong with the ili9488 ? does the touch is different ?

Here is the code of the example with pin 6 as CS

```

#include <XPT2046_Touchscreen.h>
#include <SPI.h>

#define CS_PIN  6
// MOSI=11, MISO=12, SCK=13

// The TIRQ interrupt signal must be used for this example.
#define TIRQ_PIN  3
XPT2046_Touchscreen ts(CS_PIN, TIRQ_PIN);  // Param 2 - Touch IRQ Pin - interrupt enabled polling

void setup() {
  Serial.begin(38400);
  ts.begin();
  // ts.begin(SPI1); // use alternate SPI port
  ts.setRotation(1);
  while (!Serial && (millis() <= 6000));

}

void loop() {

  if (ts.touched()) {
	TS_Point p = ts.getPoint();
	Serial.print("Pressure = ");
	Serial.print(p.z);
	Serial.print(", x = ");
	Serial.print(p.x);
	Serial.print(", y = ");
	Serial.print(p.y);
	delay(30);
	Serial.println();
  }
  delay(100);

	/*
  // tirqTouched() is much faster than touched().  For projects where other SPI chips
  // or other time sensitive tasks are added to loop(), using tirqTouched() can greatly
  // reduce the delay added to loop() when the screen has not been touched.
  if (ts.tirqTouched()) {
    if (ts.touched()) {
      TS_Point p = ts.getPoint();
      Serial.print("Pressure = ");
      Serial.print(p.z);
      Serial.print(", x = ");
      Serial.print(p.x);
      Serial.print(", y = ");
      Serial.print(p.y);
      //
      delay(30);
      Serial.println();
    }
  }*/
}
```

Thanks in advance,

koattila, 2023-09-05:
Is there any solution for this? I have the same problem, it does not work for ili9488.

PaulStoffregen, 2023-09-05:
> What's wrong with the ili9488 ? does the touch is different ?

On most displays the touch interface is made with a separate chip.  Only the company who sold this display can answer regarding which hardware it really has.

You didn't even give a link to the product or photo of the display.  How can you except anyone to be able to answer this sort of question?  Any answer with zero information about the actual display would be blind guessing!

koattila, 2023-09-05:
Hi,
I know it's just the raw display, but I put an external XPT2046 chip on the schematic (see above).
So the circuit does have a controll chip.

[image]

PaulStoffregen, 2023-09-05:
Maybe try with a board like Arduino Uno?

PaulStoffregen, 2023-09-05:
ESP has complex issues with SPI.  I do not use ESP, so can not help.  But simple boards like Uno R3 are known to work well and easy to use, with simple SPI.  Better to test with simple first.

koattila, 2023-09-05:
Okay, I give it a try, thank you!

koattila, 2023-09-05:
Before started to rewire everything, I tried this library as the last resolt, and it was working perfectly.

[Bodmer TFT touch](https://github.com/Bodmer/TFT_Touch)

just live it here in case anyone have similar problem.

gdcircuit, 2023-12-29:
I had the same issue and I believe that it was simply due to the fact that I was using a hardware uart for the display driver, and just daisy chaining the serial data to the touch chip. So whenever I would do a touch request it was using the same speed as the display was using which is too fast for the touch chip. Touch chip is around 2mhz. When you use the Bodmer lib, its using a separate soft uart, so the speed is much less. I think you can get around this by using the spi set clock command, but it works fine by just having a separate soft spi bus to the touch chip.