PaulStoffregen/XPT2046_Touchscreen #48: Ability to reassign SPI pins (MOSI,MISO,SCK)

Bwanna, 2023-07-24:
Is there a way to reassign the SPI pins?  Tried looking through the files and couldn't figure it out.

Here are the assignments I need:
MISO 12
MOSI 13
SCLK 14
DC   2
TOUCH_CS 33

Thank you ahead!

Bwanna, 2023-07-24:
Thank you Paul for the quick reply!   I could use assistance in applying this library to this device setup: [https://www.elecrow.com/esp32-display-2-4-inch-intelligent-spi-tft-lcd-touch-screen-hmi-display.html](url)

If I understand you correctly, does this library uses the SPI pins identified by the board selection?  In this case, the ESP32-WROOM-DA

Thank you

PaulStoffregen, 2023-07-24:
This issue tracker is meant only for reporting software defects.  It is not the place to ask for technical support.

closed 2023-07-24 by PaulStoffregen

Bwanna, 2023-08-09:
Paul,
How can I designate the ESP32 HSPI SPI instead of using (by default) the VSPI?

ESP32 does support a second hw SPI... Not trying to select arbitrary pins for SPI use, just want to designate the available HSPI pins.

Thanks