PaulStoffregen/XPT2046_Touchscreen #50: Platformio Release Outdated

TheNitek, 2023-11-19:
The current release with platformio is currently 4 years old (https://registry.platformio.org/libraries/paulstoffregen/XPT2046_Touchscreen ). Would it be possible to release an up to date version? (I know I can work around by using github as a dependency, but I still consider it a work around)

mattbaker-digital, 2024-10-03:
This would have been very helpful. I've just switched to a Teensy 3.6 build using a PlatformIO project and spent ages trying to figure out why the build fails with a FlexIO error. I've just tried the master branch directly and it all works now.

Bazmundi, 2024-10-22:
Use XPT2046_Touchscreen_TT, it seems to be a copy of XPT2046_Touchscreen (not forked but numbering starting at 1.5, so assume it's based upon 1.4 XPT2046_Touchscreen at your peril).

TheDevMinerTV, 2025-01-28:
6 years out of date now.