PaulStoffregen/XPT2046_Touchscreen #53: Did some one get the xp2046 to work with an ESP32-S3

ats3788, 2024-05-25:
Did some one get the xp2046 to work with an ESP32-S3
I tried 
`
#define XPT2046_IRQ 1   // T_IRQ
#define XPT2046_MOSI 42  // T_DIN
#define XPT2046_MISO 2  // T_OUT
#define XPT2046_CS 41    // T_CS
#define XPT2046_CLK 40   // T_CLK
`

with no success
Any clue ?

obanach, 2024-08-11:
```cpp
// Setup for the ESP32 S3 with ILI9341 display
// Note SPI DMA with ESP32 S3 is not currently supported
#define USER_SETUP_ID 70
// See SetupX_Template.h for all options available
#define ILI9341_2_DRIVER
// #define TFT_RGB_ORDER TFT_BGR
#define TFT_INVERSION_ON

// Typical board default pins - change to match your board
#define TFT_CS   10 //     10 or 34 (FSPI CS0) 
#define TFT_MOSI 11 //     11 or 35 (FSPI D)
#define TFT_SCLK 12 //     12 or 36 (FSPI CLK)
#define TFT_MISO 13 //     13 or 37 (FSPI Q)

// Use pins in range 0-31
#define TFT_DC    7
#define TFT_RST   -1
 
#define TFT_BL   7            // LED back-light control pin
#define TFT_BACKLIGHT_ON HIGH 

#define TOUCH_CS 16 // Optional for touch screen
#define XPT2046_IRQ 8   // T_IRQ
#define XPT2046_MISO 18  // T_OUT
#define XPT2046_MOSI 17  // T_DIN
#define XPT2046_CS 16    // T_CS
#define XPT2046_CLK 15   // T_CLK

#define LOAD_GLCD
#define LOAD_FONT2
#define LOAD_FONT4
#define LOAD_FONT6
#define LOAD_FONT7
#define LOAD_FONT8
#define LOAD_GFXFF

#define SMOOTH_FONT

#define SCREEN_WIDTH 320
#define SCREEN_HEIGHT 240
#define FONT_SIZE 4

// FSPI (or VSPI) port (SPI2) used unless following defined. HSPI port is (SPI3) on S3.
#define USE_HSPI_PORT

//#define SPI_FREQUENCY  27000000
#define SPI_FREQUENCY  40000000   // Maximum for ILI9341

#define SPI_READ_FREQUENCY  6000000 // 6 MHz is the maximum SPI read speed for the ST7789V

#define SPI_TOUCH_FREQUENCY 2500000

#define LED1_PWM 1
#define LED2_PWM 2
#define LED3_PWM 42

#define PUMP_PWM 41
#define FLOW1_SIGNAL 35
#define VALVE1_SIGNAL 36
```

pvetal, 2024-09-27:
[obanach](https://github.com/obanach) thank you very much for your solution - it was helpful for me. 
I succeeded "friendship" to ESP32 S3 with 4" display ST7796. 
I used libraries TFT_eSPI and XPT2046_Touchscreen.