PaulStoffregen/XPT2046_Touchscreen #54: The begin() function can also specify communication on a different SPI port.    ts.begin(SPI1); // use SPI1 rather than SPI is not available

ats3788, 2024-05-31:
Hello 
I fiddle with my Cheap yellow Display but I can't assign SPIClass(VSPI);
The begin() function can also specify communication on a different SPI port.
  `ts.begin(SPI1); // use SPI1 rather than SPI`

This is not available so how can I do this

```
SPIClass mySpi = SPIClass(VSPI);
XPT2046_Touchscreen ts(XPT2046_CS, XPT2046_IRQ);

TFT_eSPI tft = TFT_eSPI();
```

ats3788, 2024-05-31:
I was a bit to fast I can compile it with the Arduino IDE but not with PlatformIO any hint.
I will open a case in PlatformIO

mar-things, 2025-07-15:
I have the exact same issue here :(

chris-truter, 2025-07-27:
Try adding the lib like so in platformio.ini
`lib_deps =
    ...
    https://github.com/PaulStoffregen/XPT2046_Touchscreen.git#v1.4
    ...
`

mar-things, 2025-07-28:
Thank you Chris, indeed that was the solution. For some reason I think that platformio is fetching an older version of the library that doesn’t support giving the pins of the MCU as parameters for the screen(ts) object.